rejas/MMM-MotionDetector #1: Not working with Acer Aspire One D260 camera

rejas, 2017-11-22:
Hi rejas!, Cool idea. Which cameras will work? I'm just playing around with an old  Acer Aspire One D260. Atom CPU, 2 GiB RAM and 160 GiB HDD and Xubuntu  16.4.3 LTS. It has a buid-in webcam, which I would like to use for  motion detection. I installed the module, set the timeout to 6000 for  testing, covered the cam, but nothing happens. The module doesn't turn  off the screen.  To doublecheck, if the camera works, I installed  "Cheese" (a webcam utility) and I can take fotos and videos with the  cam. Any idea? Or will it work only with the RPi-Cam? Thanks in advance and regards from Berlin @wominator
--
What I found in the mm-out-0.log:

    magicmirror@2.1.3 start /home/wominator/MagicMirror
    sh run-start.sh

Starting MagicMirror: v2.1.3
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: MMM-TextClock.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: calendar
Initializing new module helper ...
......
Module helper loaded: MMM-MotionDetector
All module helpers loaded.
Starting server on port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: MMM-PublicTransportBerlin
Connecting socket for: MMM-DWD-WarnWeather
MMM-DWD-WarnWeather helper started...
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-Remote-Control
Starting node helper for: MMM-Remote-Control
Connecting socket for: MMM-MotionDetector
Sockets connected & modules started ...
monitor on: true
monitor :
Launching application.
Create new calendar fetcher for url: https://calendar.google.com/calendar/....
Create new news fetcher for url: http://www......
monitor :
Transportation fetcher for station Berlin, XYZ-Str. created. (Station ID: 90XYZ)
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :
monitor has been deactivated
monitor :
monitor :

It looks like, the module wants to turn of the screen, but it is switched on directly again.

Which other logs would you need?

rejas, 2017-11-22:
Does your camera show up in the `dmesg` output?
Mine shows something like this
...
[    2.564509] usb 1-1.2: New USB device found, idVendor=1415, idProduct=2000
[    2.576965] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.589306] usb 1-1.2: Product: USB Camera-B4.09.24.1
[    2.601364] usb 1-1.2: Manufacturer: OmniVision Technologies, Inc.
...

Wominator, 2017-11-23:
Hi rejas!
dmesg shows:
...
[   28.074567] uvcvideo: Found UVC 1.00 device 1.3M WebCam (0402:9665)
[   28.084375] uvcvideo 1-4:1.0: Entity type for entity Extension 6 was not initialized!
[   28.084388] uvcvideo 1-4:1.0: Entity type for entity Processing 5 was not initialized!
[   28.084396] uvcvideo 1-4:1.0: Entity type for entity Selector 4 was not initialized!
[   28.084404] uvcvideo 1-4:1.0: Entity type for entity Camera 1 was not initialized!
[   28.103012] input: 1.3M WebCam as /devices/pci0000:00/0000:00:1d.7/usb1/1-4/1-4:1.0/input/input8
[   28.104674] usbcore: registered new interface driver uvcvideo
[   28.104678] USB Video Class driver (1.1.1)
.....
Regards
wominator

rejas, 2017-11-25:
Ok, the system seems to find the camera (I guess :-) 
I added some more logging to my module, could you do a "git pull" for my repo and post the console logs from the browser you show your MagicMirror on?

Wominator, 2017-11-28:
Hi!

Sorry for being late. I was away from home for the past days. I did the update. Which Log do you mean and how can I c&p it?
Sorry for asking! I really appreciate your contribution!
Regards 
womi

rejas, 2017-12-01:
When running MM in the serveronly mode (see github main page for that) and allow your local ip access to it (ipWhitelist config parameter), you can use any browser to get the MM page. Then you can open the brwoser console and see more logs. If you still have question marks over your head, feel free to ask more :-)

ChknWing, 2018-02-23:
Rejas, I have been working on my smart mirror and included this module. Is it made for a USB camera or will Rpi camera work? Or does it matter? I currently have it running but am not getting it to sleep and wakeup my mirror. From my viewpoint, it's not doing anything. Do I need to specify what output it is looking for a signal from. BTW, I am very new to all of this. Thanks

rejas, 2018-02-26:
Hi @ChknWing I use it with my ps3 camery, havent tried it with any other camery yet. It shouldnt matter as long as the OS recognizes the camera. I didnt have to add or config anythign besides what is stated in the README here. Does the MagicMirror software start the plugin? Do any messages appear in the logs?

M1dn8, 2018-03-04:
Hey Rejas, thx for your work!
I've the same problem here too...i use a RpiCam. I could capture images and have no messages in my logs while running MagicMirror...so what could it be? Do u have an idea?

thank u 👍

rejas, 2018-03-04:
Hi @m1dn8, unfortunatly I just left for traveling for two weeks, so I dont have access to my mirror and pi to look up some stuff. 
For the time being (and before I dont have any internet acess anymore) I can only redirect you to the offical forum (https://forum.magicmirror.builders/) where you can ask for help.
I only have tried my module so far with a playstation3 toy camera and that worked out of the box.

M1dn8, 2018-03-24:
Hi @rejas, i think it doesn't work cause i have try to use the FacialRecognition together with your tool! Maybe u could find a way how we could use both together at the same time?!
thx for your work

rejas, 2018-03-25:
@M1dn8 Could you open a new issue for this interaction between FacialRecognition and my tool? And put some data on how you observed this into it? Will be closing this here since the OP @Wominator didnt follow up with more infos.

closed 2018-11-09 by rejas