rejas/MMM-MotionDetector #2: Deactivates monitor but will not wake up

shayfer, 2018-10-26:
I have a Logitech camera installed via USB that the PI recognizes. This is a fresh install with the newest build. The module blanks the screen but never wakes up. I have been looking at the logs below but do not see what is the issue.

Starting MagicMirror: v2.5.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: MMM-MotionDetector
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: MMM-MotionDetector
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
MMM-MotionDetector: monitor on true
MMM-MotionDetector: monitor state 0x120009 [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

Launching application.
MMM-MotionDetector: MOTION_DETECTED
MMM-MotionDetector: monitor state 0x120009 [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

Create new calendar fetcher for url: http://www.calendarlabs.com/templates/ical/US-Holidays.ics - Interval: 300000
Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
MMM-MotionDetector: DEACTIVATE_MONITOR
MMM-MotionDetector: monitor has been deactivated
MMM-MotionDetector: monitor state 0x120009 [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

Thanks for any help!

rejas, 2018-10-27:
Hi @shayfer can you try to activate your monitor once it blanks with this console command:
`/opt/vc/bin/tvservice --preferred && sudo chvt 6 && sudo chvt 7` ?
This is what the module uses to activate the monitor, got to find out if any error are thrown when it is used on your side...

shayfer, 2018-10-27:
rejas,

Thanks for the reply. I get this error message via console. Powering on HDMI with preferred settings
[E] Failed to power on HDMI with preferred settings

rejas, 2018-10-27:
Will look into switching the console command for turning off/on the monitor tomorrow as mentioned in #3. Stay tuned :-D
In the meantime: you are not running the mm with vc4-kms-v3d or vc4-fkms-v3d (just making sure)?

shayfer, 2018-10-27:
Thanks. Yes, I am running the default legacy video driver.

shayfer, 2018-10-27:
Re-imaged PI, selected Legacy using sudo raspi config and changed to 256 Split Memory. Works just great!  Is there any way to change it when it deactivates to go to a black screen instead of no signal?

Thanks!

rejas, 2018-10-28:
good morning @shayfer I tried succesfully at my mirror to use the vcgencmd command. Can you try out my `vcgencmd`branch and post your results here?

shayfer, 2018-10-28:
@rejas afternoon. I tried the vcgenmd branch and that works as well!  Not sure if you saw my previous comment but is there a way to go to a blank screen instead of no signal when motion is not detected? My issue is I am using a TCL 32S305 TV.  If no signal is detected it stays at a solid red screen saying no signal. There is no option to make it a black screen via the settings. (Also the screensaver option in the TV will ONLY come out of screensaver mode via TV remote and not when it discovers a signal). 

Thanks for all your help!

rejas, 2018-11-09:
hi @shayfer sorry for the late answer and for having to tell you that going to a blank screen is not possible with this repo. not sure to be honest how to make that work. is that something the monitor can do or do we need to lay over a black image over all? I am definitly interested in making the module do more but not sure how to proceed from here.

shayfer, 2018-12-01:
rejas I meant to reply earlier to this as well. I haven't made any progress on the above. I did send out a bid on freelancer.com to make but haven't had anyone bite yet.

closed 2018-12-23 by shayfer