rejas/MMM-MotionDetector #3: Use in conjunction with mmm-remote-control

adbeem, 2018-10-27:
Hi rejas!
Thank you for your awesome work as this module works perfectly for me. However, i sometimes need to update my MagicMirror through mobile and since this module doesn't work with fake kms + localhost, this conflicts with MMM-Remote-Control module as it requires the MagicMirror's address to be 0.0.0.0 and uses fake kms driver! 
Have you found any solution for this yet?

rejas, 2018-10-27:
Hi @adbeem sorry, I havent looked for a solution. Any help would of course be greatly appreciated. Do you got any ideas on this?

adbeem, 2018-10-27:
Trust me i'd love to but i'm total noob at this haha

rejas, 2018-10-27:
Darn, so I have to do something myself :-) Oh well, since you and #2 are here I might do some googling and/or testing on sunday (I am a noob in regards to the console commands for turning off/on). Maybe I can replace tvservice with something more reliable... Maybe something described here: https://raspberrypi.stackexchange.com/questions/52042/turning-tvservice-on-and-off-leaves-screen-blank

rejas, 2018-10-28:
good morning @adbeem  I tried succesfully at my mirror to use the vcgencmd command. Can you try out my `vcgencmd`branch and post your results here?

adbeem, 2018-10-28:
can you elaborate more on how to do this rejas?

rejas, 2018-10-28:
Well... Depends on where you are more proficient: Do you know git? Or do you know how to edit files on your raspeberry?

adbeem, 2018-10-28:
am i meant to edit something in node_helper.js?

adbeem, 2018-10-28:
yeb it works great but what is wrong with the old file i got it working just the same as this

adbeem, 2018-10-28:
wait rejas i'm lost lol! im not good at programming but what im saying from the beginning is your module works perfectly for me without any issues. the only issue is that i have to use 'default driver' which conflicts other module, which is mmm-remote-control (this module only works with fake kms driver)

rejas, 2018-10-31:
Ah okay.... Well, with the new branch you could use the fake-kms-driver again. But as for the 0.0.0.0 requirement, I cannot help you there. I havent found a solution for accessing the camera when not running on localhost. Maybe the author of MMM-Remote-Control has an idea @Jopyth?

adbeem, 2018-11-01:
btw rejas, sometimes the mirror wont turn back on tilll i have to restart the system. i run pm2 log and it shows below:
[TAILING] Tailing last 15 lines for [all] processes (change the value with --lines option)
/home/pi/.pm2/pm2.log last 15 lines:
PM2        | [2018-11-01T18:44:38.584Z] PM2 log: Time                 : Thu Nov 01 2018 18:44:38 GMT+0700 (Indochina Time)
PM2        | [2018-11-01T18:44:38.586Z] PM2 log: PM2 version          : 3.0.4
PM2        | [2018-11-01T18:44:38.588Z] PM2 log: Node.js version      : 10.9.0
PM2        | [2018-11-01T18:44:38.590Z] PM2 log: Current arch         : arm
PM2        | [2018-11-01T18:44:38.592Z] PM2 log: PM2 home             : /home/pi/.pm2
PM2        | [2018-11-01T18:44:38.593Z] PM2 log: PM2 PID file         : /home/pi/.pm2/pm2.pid
PM2        | [2018-11-01T18:44:38.595Z] PM2 log: RPC socket file      : /home/pi/.pm2/rpc.sock
PM2        | [2018-11-01T18:44:38.597Z] PM2 log: BUS socket file      : /home/pi/.pm2/pub.sock
PM2        | [2018-11-01T18:44:38.599Z] PM2 log: Application log path : /home/pi/.pm2/logs
PM2        | [2018-11-01T18:44:38.600Z] PM2 log: Process dump file    : /home/pi/.pm2/dump.pm2
PM2        | [2018-11-01T18:44:38.601Z] PM2 log: Concurrent actions   : 2
PM2        | [2018-11-01T18:44:38.603Z] PM2 log: SIGTERM timeout      : 1600
PM2        | [2018-11-01T18:44:38.605Z] PM2 log: ===============================================================================
PM2        | [2018-11-01T18:44:38.865Z] PM2 log: Starting execution sequence in -fork mode- for app name:mm id:0
PM2        | [2018-11-01T18:44:38.910Z] PM2 log: App name:mm id:0 online

/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | MMM-MotionDetector: MOTION_DETECTED
0|mm       | MMM-MotionDetector: monitor has been activated
0|mm       | MMM-MotionDetector: monitor display_power=0
0|mm       | 
0|mm       | MMM-MotionDetector: DEACTIVATE_MONITOR
0|mm       | MMM-MotionDetector: monitor has been deactivated
0|mm       | MMM-MotionDetector: monitor display_power=1
0|mm       | 
0|mm       | MMM-MotionDetector: MOTION_DETECTED
0|mm       | MMM-MotionDetector: monitor has been activated
0|mm       | MMM-MotionDetector: monitor display_power=0
0|mm       | 
0|mm       | MMM-MotionDetector: MOTION_DETECTED
0|mm       | MMM-MotionDetector: monitor display_power=1
0|mm       | 

/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/MichMich/MagicMirror.git/': SSL connection timeout
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/KirAsh4/calendar_monthly/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/fewieden/MMM-ip.git/': SSL connection timeout
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/ianperrin/MMM-ModuleScheduler.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/rejas/MMM-MotionDetector.git/': SSL connection timeout
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/MichMich/mmm-systemtemperature.git/': SSL connection timeout
0|mm       | 
0|mm       | fatal: unable to access 'https://github.com/sheyabernstein/MMM-connection-status.git/': SSL connection timeout
0|mm       |

rejas, 2018-11-01:
Nothing I can do about those ssl errors, maybe your internet connection had troubles in the way. 
I merged my changes into the master branch now with small modifications. So you can reinstall my module and see if that helps

adbeem, 2018-11-01:
alright i will test it for a day and see how it goes

adbeem, 2018-11-01:
yea rejas i still got the same issue :(

adbeem, 2018-11-04:
i place my mirror next to the wifi modem so i don't think my connection is affected. Can you post the old node_helper.js file so i can run some test?

adbeem, 2018-11-04:
oh nvm rejas, it's my mirror that causing the issue. screen freezes sometimes probably due to high electron usage

rejas, 2018-11-04:
Good to here :-) Have you switched drivers to maybe get the usage down? Closing this ticket nonetheless

closed 2018-11-04 by rejas