rejas/MMM-MotionDetector #6: Not working and console errors

pmishev, 2018-11-04:
It doesn't seem to work (Raspbian Stretch).

Once, MagicMirror starts, I have this output:
```
MMM-MotionDetector: MOTION_DETECTED
MMM-MotionDetector: monitor display_power=1
```
and no further output from the module appears regardless of any motion.

This is my configuration:
```
		{
		        module: 'MMM-MotionDetector',
		        config: {
		                captureIntervalTime: 1000,
		                scoreThreshold: 200,
		                timeout: 3000
		        }
		}
```

What I get in the browser console is this error:
[image]

rejas, 2018-11-05:
Hi @pmishev what browser are you using in the screenshot you posted?

pmishev, 2018-11-05:
If I go in the "What's new" section of the developer tools, I can see `Highlights from the Chrome 61 update`, so presumably that's the browser version.

rejas, 2018-11-09:
Hi @pmishev can you try my branch "gum" of this repo where I switched to a newer brwoser-api for geting the camera image and post your results here?

pmishev, 2018-11-10:
Sure thing.. I get:

```
MMM-MotionDetector: error init cam engine
initErrorCallback @ MMM-MotionDetector.js:68
```

I set a breakpoint there to see the actual error message and it is:
`Only secure origins are allowed (see: https://goo.gl/Y0ZkNV).`

This might somehow be related to an earlier warning I have in the console:
```
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:128 Electron Security Warning (Insecure Resources) This renderer process loads resources using insecure protocols. This exposes users of this app to unnecessary security risks. Consider loading the following resources over HTTPS or FTPS. 
- http://0.0.0.0:8080/css/main.css
- http://0.0.0.0:8080/fonts/roboto.css
- http://0.0.0.0:8080/socket.io/socket.io.js
- http://0.0.0.0:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js
- http://0.0.0.0:8080/js/defaults.js
- http://0.0.0.0:8080/config/config.js
- http://0.0.0.0:8080/vendor/vendor.js
- http://0.0.0.0:8080/modules/default/defaultmodules.js
- http://0.0.0.0:8080/js/logger.js
- http://0.0.0.0:8080/translations/translations.js
- http://0.0.0.0:8080/js/translator.js
- http://0.0.0.0:8080/js/class.js
- http://0.0.0.0:8080/js/module.js
- http://0.0.0.0:8080/js/loader.js
- http://0.0.0.0:8080/js/socketclient.js
- http://0.0.0.0:8080/js/main.js
- http://0.0.0.0:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
- http://0.0.0.0:8080/translations/en.json
- http://0.0.0.0:8080/translations/en.json
- http://0.0.0.0:8080/modules/default/alert//alert.js
- http://0.0.0.0:8080/modules/default/alert/classie.js
- http://0.0.0.0:8080/modules/default/alert/modernizr.custom.js
- http://0.0.0.0:8080/modules/default/alert/notificationFx.js
- http://0.0.0.0:8080/modules/default/alert/ns-default.css

For more information and help, consult https://electronjs.org/docs/tutorial/security.
This warning will not show up once the app is packaged.
warnAboutInsecureResources @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:128
loadHandler @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/init.js:212
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:188 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security Policy set or a policy with "unsafe-eval" enabled. This exposes users of this app to unnecessary security risks.

For more information and help, consult https://electronjs.org/docs/tutorial/security.
This warning will not show up once the app is packaged.
```

rejas, 2018-11-12:
Looks like you configured your mirror to run on ip 0.0.0.0 
Unfortunatly, using the webcam is only possible when running on "localhost" and on secured (httpS) hosts.

pmishev, 2018-11-13:
I had set it to 0.0.0.0 to make `MMM-Remote-Control` work, but I changed it now to `localhost` and the result is still the same in both `master` and `gum`:

```
Electron Security Warning (Insecure Resources) This renderer process loads resources using insecure protocols. This exposes users of this app to unnecessary security risks. Consider loading the following resources over HTTPS or FTPS. 
- http://localhost:8080/css/main.css
- http://localhost:8080/fonts/roboto.css
- http://localhost:8080/socket.io/socket.io.js
- http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js
- http://localhost:8080/js/defaults.js
- http://localhost:8080/config/config.js
- http://localhost:8080/vendor/vendor.js
- http://localhost:8080/modules/default/defaultmodules.js
- http://localhost:8080/js/logger.js
- http://localhost:8080/translations/translations.js
- http://localhost:8080/js/translator.js
- http://localhost:8080/js/class.js
- http://localhost:8080/js/module.js
- http://localhost:8080/js/loader.js
- http://localhost:8080/js/socketclient.js
- http://localhost:8080/js/main.js
- http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
- http://localhost:8080/translations/en.json
- http://localhost:8080/translations/en.json
- http://localhost:8080/modules/default/alert//alert.js
- http://localhost:8080/modules/default/alert/classie.js
- http://localhost:8080/modules/default/alert/modernizr.custom.js
- http://localhost:8080/modules/default/alert/notificationFx.js
- http://localhost:8080/modules/default/alert/ns-default.css

For more information and help, consult https://electronjs.org/docs/tutorial/security.
This warning will not show up once the app is packaged.
warnAboutInsecureResources @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:128
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:188 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security Policy set or a policy with "unsafe-eval" enabled. This exposes users of this app to unnecessary security risks.

For more information and help, consult https://electronjs.org/docs/tutorial/security.
This warning will not show up once the app is packaged.
```

Maerchensarg, 2018-12-20:
I get the same error. I have set up a new RPi3B+ and camera and installed MagicMirror. This is the first module and I get the same error as pmishev wrote at the beginning.

MCGregorUK, 2019-02-20:
First: Thanks for your Module!
I had the same problem and fixed it.
You have to remove or change the ip-address under
```
var config = {
//    	address : '0.0.0.0',
```
I comment it out...

rejas, 2019-02-28:
Thx for the info @MCGregorUK Added your suggestion to the README. Closing this now. If the OP still has problems he should come back and re-open it.

closed 2019-02-28 by rejas

djay07, 2019-06-02:
> I had set it to 0.0.0.0 to make `MMM-Remote-Control` work, but I changed it now to `localhost` and the result is still the same in both `master` and `gum`:
>
> ```
> Electron Security Warning (Insecure [...]

Did you make it work with MMM-Remote-Control in parallel?

rejas, 2019-06-02:
Not sure what this log is, but I dont see any mentioning of my module there...

djay07, 2019-06-03:
Well I made your module work too changing address from 0.0.0.0 to localhost.
This breakes the remote control module. 
The guy i quoted had the same issue. I was wondering if he was able to solve it?

Do u have an idea how to make it work on 0.0.0.0 as adress value?

rejas, 2019-06-20:
Had a look at ti and still, it isnt possible to use adress 0.0.0.0 with my module due to https://stackoverflow.com/questions/34215937/getusermedia-not-supported-in-chrome

maobendorfer, 2019-11-24:
Hi,
you could solve the problem with the 2 modules needing different configs by setting the config to localhost and running a proxy which takes calls against an IP and forwards them to localhost. There are several solutions out there, but one simple might be using an npm proxy module for the express server and starting it via pm2 like the MM.
As a further step you might want to add some credentials to the proxy as you would somhow bypass browser security.
If someone needs an example, let me know and I'll post one when I'm back at my desktop.

rejas, 2019-11-30:
HI @bavario would be awesome to have an example from you in the README for this configuration problem. Are you already back at your desktop for a PR ? :-)

ngdemigod, 2020-09-12:
@bavarioWould you be able to show how to config the proxy server? I would like to have parallel access to both MotionDetector & Remote Control but I'm confused on how to set it up