rejas/MMM-MotionDetector #8: How can I get this working with the wired picamera?

mutemute, 2019-02-04:
Module runs fine, but there's no activity led on the camera, so it's not being accessed.. thank you!

rejas, 2019-03-12:
This module here works by using the navigator.mediaDevices.getUserMedia functionality of the browser / electron. I assume this isnt possible with the raspi cam out of the box. 
Unfortunatly my interaction possibilities are limited ATM. Maybe you can try this out https://reprage.com/post/pi-camera-module-in-the-browser and see if it shows up then?

adamlove86, 2019-03-13:
Not sure if I did the right thing but I added module MMM-iFrame. First up tried just putting camera.html locally on the pi, and I'd get "Cannot GET [local path]/camera.html". The camera.html file being the page outlined on https://reprage.com/post/pi-camera-module-in-the-browser.

Then tried putting camera.html on http://[another mini server on my home network]/camera.html. 
Both times the space for the iFrame would load, but no picture was coming up. 2nd time no error though. Doesn't appear to be activating the camera (no red light either).

I'm not sure below link is of any assistance to point me in the right direction? They also refer to getUserMedia
https://github.com/raspberrypi/linux/issues/1498

rejas, 2019-03-27:
Re-openign so I might have a look at it later. Got no pi-camera though yet.

rev138, 2019-04-15:
1. Open /etc/modules-load.d/modules.conf
2. Add `bcm2835-v4l2` to the end of the file and save it.
3. Reboot.
4. Profit

rejas, 2019-04-15:
So you got my module working with the pi camera @rev138 ?

rev138, 2019-04-15:
@rejas I did, yes. It works fine, you just need to load the kernel module for the video.

rejas, 2019-04-16:
Thx for the info @rev138 Is that module already installed in the standard raspberry image? WOndering if those 4 steps are all that is needed so I can add it to the README for other users to follow.

rev138, 2019-04-16:
It should be, yes.

rejas, 2019-04-16:
Add your tips to the README!

closed 2019-04-16 by rejas

rev138, 2019-04-16:
Thanks for making this module!

rejas, 2019-04-16:
Thanks to https://github.com/alexyak :-) I just polished it :-)

rejas, 2025-11-16:
hi @laveshnk-crypto thanks for trying out my module.

I recently updated the module, and tested it with the latest raspiOS ("trixie"), so I probably need to update the README:
- my ps3 usb cam worked out of the box, I only need to config my module for using the trixie-tools (see new config option "platform")

If you could test your rpi-camera 3 with the latest version from master, this would be really helpful