rejas/MMM-MotionDetector #17: It works.....but after a while entire pi goes to some kind of sleep?

dxfan227, 2019-04-20:
I have the module up and running properly ( I think) I am using the PSEEYE camera as indicated in your readme.md. everything works pretty ok for the most part....screen turns off, motion, screen turns on. 

I tried it for an extended period of time though and when I woke up this morning I couldn't get the screen to come on, the red light on the camera ( indicating watching) is also off. Even manually turning the screen on I have no signal....and trying to use the keyboard and mouse doesn't wake it up.

The pi is powered on still...and when I ping it it is accessible and I can SSH into it properly. 

Do you haver any ideas?

rejas, 2019-04-21:
Did you have a look at the logs of the magicmirror to see if there are any errors @dxfan227 ?

dxfan227, 2019-04-23:
Was able to get this working, thank you.

closed 2019-04-23 by dxfan227

rejas, 2019-04-23:
What was your fix? Maybe its helpful for others?

dxfan227, 2019-04-23:
sure, I feel stupid.

So I was doing all my testing/file edition via ssh on my Mac. I would npm start magic mirror via command line and then close my Mac and go to bed, only to wake up to the pi unresponsive. I didn't realize that terminating the CLI from the Mac would terminate the program as well.

It was left in a closed state with the monitor stuff in an " off state" until I manually sent the command. 

the fix was making sure magic mirror was being launched from the pi itself.