rejas/MMM-MotionDetector #18: can this module send our a custom notification?

dxfan227, 2019-04-23:
Was hoping to sync this with another module to refresh upon wake/motion

rejas, 2019-04-23:
Its already sending notifications, cant you handle those in our other modules?

dxfan227, 2019-04-23:
probably. I don't know any coding what so ever. what notification does the module broad cast when it detects motion?

rejas, 2019-04-23:
When detected its sends MOTION_DETECTED and DEACTIVATE_MONITOR if the monitor gets back to sleep

dxfan227, 2019-04-23:
@rejas  thanks...i'll see if I can use this. 

unfortuantly after I closed the other issue thread I went to my mirror and mmm-motion detector was not active again.. I manally sent the screen on command via terminal and when the screen loaded magic mirror was still running, so this time there was something strange about the module crashing or something. I'll keep an eye on it, in the mean time is there any good way to check a log when something like this happens? something that would give me info from hours ago.

closed 2019-04-24 by dxfan227