rejas/MMM-MotionDetector #19: Gets deactivated somehow

dxfan227, 2019-04-24:
Hi again, sorry to be such a bother. So I came home again tonight and MMM-Motion-Detector was not active ( no red light on the PSeye camera). After manually sending the screen on command I was looking at my MagicMirror screen.

Is there anyway to check a log or something to see how/when/why mmm-motiondetector stopped responding? I tired looking at pm2 logs but wasn't really sure about it. 

I did notice one thing, every time mmm-notification sends a notification to mmm-googleroute to refresh.....the camera " detected motion"

rejas, 2019-04-25:
so "pm2 log <name>" doesnt work for you?

dxfan227, 2019-04-25:
I looked through the pm2, pm2 error mm, and pm2 out mm. There are no errors or crashes reported nothing wrong with mm-motion detector.... it doesn’t show the module “quitting”

But still after 10 min of monitor off the red light in the pseye camera turns off... blue light still on so it’s recieving power.

rejas, 2019-04-25:
My module doesnt control the camera itself, it just reads it input via the browser. It cannot send it to sleep or anything like that. The OS probably does that turning off you describe so thats why there are no infos in the log. Sorry, cant help you here.

dxfan227, 2019-04-25:
Hmm I am using raspberry pi raspbian stretch. I wonder what is causing that. 

Any ideas at all?

dxfan227, 2019-04-25:
The camera only turns on when the module is run.... so what is telling the camera to turn on?

rejas, 2019-04-25:
The browser (in this case Electron) is handling the camera. Aure you using the latest version of raspbian and magicmirror?

dxfan227, 2019-04-25:
I do not have the latest version of magic mirror ( the one that was just released) because people where having issues. Do u think I need to upgrade?

rejas, 2019-04-26:
You could try it (going back to an old version is always possible). I have a better experince with the latest version (while older versions had some hangups), but your mileage may vary.

SamerAl-khateeb, 2019-05-15:
Sorry if this question is not a smart question!. If I understand correctly, this module should turn the magic mirror screen/monitor on or off if there is motion detected, am I correct? does it really switch it off or does it make the monitor go black only?

dxfan227, 2019-05-15:
It really switches it off.

Thanks,

David B. Ruiz |
*Director of Recruitment**G*
*lobal Talent Solutions, Inc.**Email:* druiz@gtsrecruiters.com
*Office:* 714-921-6428 x. 307
*Mobile:* (408) 641-9701
*Web:* http://www.GTSrecruiters.com <http://www.gtsrecruiters.com/>

SamerAl-khateeb, 2019-05-15:
@dxfan227  this is great! thank you. I just installed it and it does work very well with the PlayStationEye!

rejas, 2019-05-16:
Happy to hear this :-)

closed 2019-05-16 by rejas