rejas/MMM-MotionDetector #21: MotionDetector + Facial recognition

F17MC, 2019-06-12:
Hello Rejas! 
I'm using your module and it is fabulous!

Now, I want to add a facial recognition.
I have a python script, witch verify what person are in front of my mirror.
The script works fine, and I want add it to the MotionDetector module.

My idea is:
Detection motion
turn on the screen and run the script
if script return >75 (% of recognition)
exec notification to show more modules
else break, continue or nothing

Can you help me with this feature? Thanks!

rejas, 2019-06-12:
Hi F17MC 
the idea sounds nice but I go a few questions: 
- how is the recognition done? 
- is some external service used?
- what happens with the data from the camera? (privacy)
- is the code currently somewhere visible for me to take alook at?
Happy to hear back from you

gazambuja, 2019-06-20:
You need to modify the code from `if (score > this.config.scoreThreshold)` change to something like:
`if (score > this.config.scoreThreshold && recognitionScore > 75 )`

And of course, set the recognitionScore var before that with the recognition result.

rejas, 2019-06-21:
@gazambuja what is the purpose of your comment?

closed 2020-04-08 by rejas