rejas/MMM-MotionDetector #22: Monitor stays on

SdeGeata, 2019-06-22:
I just installed your module, used the config as is from the read me. Modified for PiCam and rebooted, but nothing happens - the monitor just stays on. Any suggestions?

rejas, 2019-06-22:
did you turn on the UI of the module? If yes, did it measure any score? can you see if there are any errors on the console?

SdeGeata, 2019-06-24:
When you say "turn on" do you mean drop the module in the config.js? If so, yes. I see the debug output on the screen - nothing important it would seem though. It looks like the screen-shot attached, and output never changes.

[image]

rejas, 2019-06-24:
okay, so the module is loaded :-) have you followed the steps in the README on how to enable the pi cam ?

SdeGeata, 2019-06-24:
Yup, did that too.
[image]

rejas, 2019-06-24:
Can you [enable the developer tools ](https://discuss.atom.io/t/how-to-make-developer-tools-appear/16232 )on your electron instance and look for errors ?

SdeGeata, 2019-06-24:
Electron?

rejas, 2019-06-24:
Thats the UI Window MagicMirror runs in :-)

SdeGeata, 2019-06-24:
The error found in Developer tools:

`MMM-MotionDetector.js:76 MMM-MotionDetector: DiffCamEngine init failed. NotReadableError: Could not start video source
initErrorCallback @ MMM-MotionDetector.js:76

`

rejas, 2019-06-24:
Last question :-) What adress is your MM running on?

SdeGeata, 2019-06-24:
192.168.1.108

rejas, 2019-06-24:
Thats the one in your config.JS? Because my module only runs in localhost or https URLs since Elektron demands that for camera access

SdeGeata, 2019-06-24:
Ah, gimme a sec...

yup, set to localhost.

rejas, 2019-06-24:
Grmpf..  Now I am stuck, since I dont habe a pi cam to actually test my module against. Could only add more debugging to it. Could you post your whole console output to this ticket? Will take a look at it later...

SdeGeata, 2019-06-24:
This is all of it from "errors & warnings";

`/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:127 Electron Security Warning (Insecure Resources) This renderer process loads resources using insecure
    protocols.This exposes users of this app to unnecessary security risks.
    Consider loading the following resources over HTTPS or FTPS. 
 - http://localhost:8080/css/main.css
- http://localhost:8080/fonts/roboto.css
- http://localhost:8080/socket.io/socket.io.js
- http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js
- http://localhost:8080/js/defaults.js
- http://localhost:8080/config/config.js
- http://localhost:8080/vendor/vendor.js
- http://localhost:8080/modules/default/defaultmodules.js
- http://localhost:8080/js/logger.js
- http://localhost:8080/translations/translations.js
- http://localhost:8080/js/translator.js
- http://localhost:8080/js/class.js
- http://localhost:8080/js/module.js
- http://localhost:8080/js/loader.js
- http://localhost:8080/js/socketclient.js
- http://localhost:8080/js/main.js
- http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
- http://localhost:8080/translations/en.json
- http://localhost:8080/translations/en.json
- http://localhost:8080/modules/MMM-MotionDetector//MMM-MotionDetector.js
- http://localhost:8080/vendor/node_modules/moment/min/moment-with-locales.js
- http://localhost:8080/modules/MMM-MotionDetector/diff-cam-engine.js
- http://localhost:8080/modules/MMM-MotionEye//MMM-MotionEye.js
- http://localhost:8080/modules/default/clock//clock.js
- http://localhost:8080/vendor/node_modules/moment-timezone/builds/moment-timezone-with-data.js
- http://localhost:8080/modules/default/clock/clock_styles.css
- http://localhost:8080/modules/default/calendar//calendar.js
- http://localhost:8080/modules/default/calendar/calendar.css
- http://localhost:8080/vendor/css/font-awesome.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/all.min.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/v4-shims.min.css
- http://localhost:8080/modules/MMM-MyWastePickup//MMM-MyWastePickup.js
- http://localhost:8080/modules/MMM-MyWastePickup/MMM-MyWastePickup.css
- http://localhost:8080/modules/mmm-moon-phases//mmm-moon-phases.js
- http://localhost:8080/modules/MMM-bitcoin//MMM-bitcoin.js
- http://localhost:8080/modules/MMM-bitcoin/MMM-bitcoin.css
- http://localhost:8080/modules/mmm-udr//mmm-udr.js
- http://localhost:8080/modules/MMM-DarkSkyForecast//MMM-DarkSkyForecast.js
- http://localhost:8080/modules/MMM-DarkSkyForecast/skycons.js
- http://localhost:8080/modules/MMM-DarkSkyForecast/MMM-DarkSkyForecast.css
- http://localhost:8080/modules/MMM-Parcel//MMM-Parcel.js
- http://localhost:8080/modules/MMM-Parcel/MMM-Parcel.css
- http://localhost:8080/modules/MMM-Strava//MMM-Strava.js
- http://localhost:8080/modules/MMM-Strava/MMM-Strava.css
    
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
warnAboutInsecureResources @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:127
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:184 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
      Policy set or a policy with "unsafe-eval" enabled. This exposes users of
      this app to unnecessary security risks.
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
warnAboutInsecureCSP @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:184
undefined:1 Failed to load resource: the server responded with a status of 404 (Not Found)
MMM-MotionDetector.js:76 MMM-MotionDetector: DiffCamEngine init failed. NotReadableError: Could not start video source
initErrorCallback @ MMM-MotionDetector.js:76`

rejas, 2019-06-24:
Got to admit that I cant help you from here, that pi cam doesnt seem to work. Maybe @rev138 who provided the original fix for the pi cam can help here?

rev138, 2019-06-24:
@SdeGeata : Can you run this is and post the output?

`sudo lsmod | grep bcm2835-v4l2`

rev138, 2019-06-24:
Another thought -- make sure the camera is connected to the right port. There are two identical looking connectors on the board. One's for the camera, the other isn't. I've made that mistake :)

SdeGeata, 2019-06-25:
Hello rev138;

I ran your command line, no output in terminal and nothing changed in MM. Same output from the developer tools:

`Failed to load resource: the server responded with a status of 404 (Not Found)
MMM-MotionDetector.js:76 MMM-MotionDetector: DiffCamEngine init failed. NotReadableError: Could not start video source
initErrorCallback @ MMM-MotionDetector.js:76`

As far as the camera port; I am indeed connected to the correct one. It is already bing used effectively for MotionEye.

SdeGeata, 2019-06-25:
I appear to have gotten it to work! I am troubleshooting some issues with MMM-Facial-Recognition-OCV3 and I turned off the Motion-Detection module as well as MotionEye to see what would happen. No change in the one I was troubleshooting for, but i turned back of MMM-MotionDetector and suddenly got output on the debug. I waited 2 minutes and the monitor shut off!

So it would seem I can't use the same camera for MMM-MotionDetector and MotionEye? Do you think there's a work-around for this conflict, or should I just get a second camera module?

SdeGeata, 2019-06-26:
I would like to know however, how do I hide the debug info now while the module is running?

rejas, 2019-10-09:
Sorry for being not responsive @SdeGeata your issue got buried :-( 
As for the conflict between this module and motioneye I cant help you with any new ideas
To hide the debug info, you just have to remove the line "position: ...." you have in your config under the MMM-MotionDetector settings.

airart, 2020-04-07:
great job, I wanted to know if it's possible to go to sleep with a clock module and not turn off the monitor. the final goal is to put a clock module as a screensaver and wake up the MM with the ps3cam that works well

rejas, 2020-04-08:
Thx for the praise @airart BUt I dont understand your question or what you want to do. Could you open new issue for that and rephrase it please?

closed 2020-04-08 by rejas