rejas/MMM-MotionDetector #25: I got errors when i try to make a pull

el97, 2019-10-28:
I got some errors i cant get rid of while im trying to update the module.
`error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.
`
And when i try to stash with the command git stash i got this error
` package-lock.json: needs merge
package.json: needs merge
package-lock.json: needs merge
package.json: needs merge
package-lock.json: unmerged (138c0d54d2895c9efc141d82345a2bd6fd0220d9)
package-lock.json: unmerged (afe6a50cfcefb524404b9841c04008f34fe240f6)
package.json: unmerged (d0322f03a60de3da38074e6e065bd72d6bcf24bf)
package.json: unmerged (0a749534eefef45cdce68020289d1979c982654c)
fatal: git-write-tree: error building trees
Cannot save the current index state
`
I would like some help with this.

rejas, 2019-10-29:
Try this command to reset your state:
`git checkout .`
Did that help?

el97, 2019-10-29:
That ended up with the following error:
`pi@MagicMirror:~/MagicMirror/modules/MMM-MotionDetector $ git checkout .
error: path 'package-lock.json' is unmerged
error: path 'package.json' is unmerged
`

rejas, 2019-11-03:
Sorry, a little sick right now and busy with the offspring, cant help you much. Please search for yourself how to reset a git repo to the latest master. Closing this since it isnt an issue of the module.

closed 2019-11-03 by rejas