rejas/MMM-MotionDetector #29: Change motion detector action

airart, 2020-04-08:
Sorry for my bad english 
Hello , is it possible to modify the module and make it display only a specified module when the cam detect a motion ?
It’s useful now but it turn off the TV and I’m forced to turn on manually After 5 minutes of inactivity 
Ma goal is use your module to switch to a full screen clock to a normal MM 
Tanks u so much

rejas, 2020-04-14:
hi @airart sorry for the late answer.
yes it would be possible to modifiy the module so that it does something similar to what you want. 
The module already sends a "MOTION_DETECTED" notification to the system so any other module could listen to that and react accordingly. I dont know if there are modules out there that switch displayed modules depneding on notifications but I would be surprised if not. 
My module "only" needs to be changed so that one can disable turning off/on the monitor autoamtically. Maybe I find some time for that later this week....

BrianHepler, 2020-06-16:
Wait, scratch that. Motion detection works without the module being given a position. My bad. It just wasn't as chatty in the logs.

rejas, 2020-07-07:
Hey @airart I just added an option to the module so that the monitor is never turned off (by setting the timeout in the config to -1) Could you check if that would help you?

BrianHepler, 2020-07-13:
Sorry, I just saw this. I'll check this out and get back to soon.

BrianHepler, 2020-07-14:
My testing is failing. After setting the timeout to -1, the module does not detection motion and never sends the notification. It also never blanks the screen, so there's that bit of good news.

The last I see in the electron console is "DiffCamEngine init successful." so that's encouraging. The stdout shows "MMM-MotionDetector: monitor display_power=1". But it doesn't respond to motion.

rejas, 2020-07-14:
Thx for testing. After looking again at my code I found some bugs in my line of thought :-)

Could you checkout the develop branch and see if now MOTION_DETECTED notifications can be received?

BrianHepler, 2020-07-15:
I switched to the devlop branch and gave it ago. Sorry, but no change in behavior.

rejas, 2020-07-15:
Question on your use case: How do you consume the notifications (aka how do you debug if my module sends the MOTION_DETECTED notifications)?

BrianHepler, 2020-07-16:
I normally put the Pi into debug mode with `npm start dev`and then monitor the logs via `pm2 log 0` to catch both electron and node_helper.js console output. If I am troubleshooting, I will add a debug switch and then write out to electron.

I have a monitor hooked up the Pi, so I can see if/when the monitor turns off.

BrianHepler, 2020-07-16:
I'm going to take a crack at it and I'll send you a PR with my take. We'll get this knocked out.

BrianHepler, 2020-07-16:
PR submitted.

airart, 2020-07-18:
Right soon as possible
Tnx 

Filippo Gallerani