rejas/MMM-MotionDetector #31: Testing results

BrianHepler, 2020-06-16:
First, thank you for the module!
Second, I just tested this on a RPi4 4GB and the RpiCam. Worked great. I had to fiddle around with the address & IP whitelist, but I got it working without having to do the proxy thing.

I did find that if you do not put a location for the module, it will not detect motion and it also will not turn off the monitor. I think having this run in the background instead of always showing debug information would be ideal.

I second the request to customize the notification send on motion detection.

rejas, 2020-07-07:
Hey @BrianHepler I added an option to never disable the monitor (by setting the timeout to -1). Could you check if that stills sends the notification and therefor would help you?

BrianHepler, 2020-07-13:
I'll check it out in the next day or so. Thank you!

BrianHepler, 2020-07-16:
Let's combine the new features into the other issue. Easier to track.

closed 2020-07-16 by BrianHepler