rejas/MMM-MotionDetector #39: iMac monitor not shutting off

enuzzo, 2020-11-09:
Hello there, we've installed Magic Mirror on an old iMac 2007 with Ubuntu in our design studio, as a smart screen companion for colleagues. Your module is working perfectly with the cam, detecting motion. Although the screen never turns off. 
According to what I red on screen, I presume it thinks to have shut it of .. but it remains awake.
Am I missing something? Can I help troubleshooting the issue?

Stay safe!
Luca C.

rejas, 2020-11-09:
The modules uses the linux command "vcgencmd". So you should first check if that is installed (or if it even is available for mac/ubuntu). In a normal linux it is already included....

enuzzo, 2020-11-10:
> Reply: rejas/MMM-MotionDetector <reply@reply.github.com>

Many thanks for your answer ;-) 
Can’t find anything similar on Ubuntu. I’ll try with a raspberry project and an attached monitor.

closed 2020-11-11 by rejas