rejas/MMM-MotionDetector #40: Issue with Mas

3vidar, 2020-12-07:
The old version worked perfectly on my Mac mini with Catalina. After the update the monitor is not turned off anymore. I see the motion in the Display debug information UI and don't get any error messages, but the monitor stays on. The funny thing is that it shows the off time correctly in the UI even though the monitor is always on.
My changes as it used to work on the Mac with the Old version:

rejas, 2020-12-07:
HI @3vidar what do you mean by "old version" ? I never had this module working for macOS so I am baffled on what version you were running?

3vidar, 2020-12-09:
Hi @rejas sorry for my late replay. I have tried to find the error but without success. But what I noticed is that it does not work with the new MagicMirror version. I have now installed my old backup and now it works again. I have the MagicMirror version 2.11.0 and MMM-MotionDetector V1.5.0 and it works perfectly, but unfortunately not with the new versions.

rejas, 2020-12-09:
do you use a browser on your mac mini to display the MM or are you using the electron version?

3vidar, 2020-12-09:
I use the electron version

rejas, 2020-12-09:
And you are on the latest master branch of the MM and my module?

3vidar, 2020-12-09:
Yes, if I update to the latest branch it won‘t work any more! Just If I stay on the old version it works perfectly.

rejas, 2020-12-09:
The latest master branch of MM or my module? Sorry for the questions, hard to debug from afar. Can you see errors on the electron console?

3vidar, 2020-12-09:
If both are on the latest version it didn‘t work and there are then no errrors! What I can do is to update your module and stay on the old MM version to see if that will work.

3vidar, 2020-12-09:
So I updated your Module to the latest version. Now my screen didn't go to sleep I I got this error message:

`[2020-12-09 22:17:58.647] [LOG]    Launching application.
[2020-12-09 22:17:58.940] [LOG]    Whoops! There was an uncaught exception...
[2020-12-09 22:17:58.940] [ERROR]  TypeError: Log.info is not a function
    at /Users/sladak/MagicMirror/modules/MMM-MotionDetector/node_helper.js:47:11
    at ChildProcess.exithandler (child_process.js:294:7)
    at ChildProcess.emit (events.js:200:13)
    at maybeClose (internal/child_process.js:1021:16)
    at Socket.<anonymous> (internal/child_process.js:430:11)
    at Socket.emit (events.js:200:13)
    at Pipe.<anonymous> (net.js:586:12)
[2020-12-09 22:17:58.944] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-12-09 22:17:58.944] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-12-09 22:18:00.170] [LOG]    Whoops! There was an uncaught exception...
[2020-12-09 22:18:00.171] [ERROR]  TypeError: Log.info is not a function
    at Class.socketNotificationReceived (/Users/sladak/MagicMirror/modules/MMM-MotionDetector/node_helper.js:54:11)
    at Socket.<anonymous> (/Users/sladak/MagicMirror/js/node_helper.js:113:11)
    at Socket.emit (events.js:200:13)
    at /Users/sladak/MagicMirror/node_modules/socket.io/lib/socket.js:528:12
    at processTicksAndRejections (internal/process/task_queues.js:82:9)
[2020-12-09 22:18:00.171] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
`

rejas, 2020-12-09:
Looks like it is not finding the "Log.info" method. But that is in MM v2.13.0 and in tge master branch of my module.

3vidar, 2020-12-09:
So I updated everything to the newest branch. Now i got his in my console:

`[2020-12-09 22:42:04.476] [INFO]   MMM-MotionDetector: monitor status is 4

[2020-12-09 22:42:04.477] [INFO]   MMM-MotionDetector: monitor is OFF.
[2020-12-09 22:42:05.963] [INFO]   MMM-MotionDetector: activating monitor.
[2020-12-09 22:42:06.013] [LOG]    Create new calendar fetcher for url: ........
[2020-12-09 22:42:06.019] [LOG]    Create new calendar fetcher for url: ......
[2020-12-09 22:42:06.020] [LOG]    Create new calendar fetcher for url: .....
[2020-12-09 22:42:06.021] [LOG]    Wed Dec 09 2020 22:42:06 GMT+0100 (Mitteleuropäische Normalzeit): Triggering Xiaomi Gateway connect
[2020-12-09 22:42:06.069] [INFO]   MMM-MotionDetector: monitor status is 4

[2020-12-09 22:42:07.190] [INFO]   MMM-MotionDetector: monitor has been activated.`

no error but my screen won't go any more to sleep

3vidar, 2020-12-09:
Log:

MMM-MotionDetector: starting up
clock.js:46 Starting module: clock
2MMM-SwissCommute.js:39 Starting module: MMM-SwissCommute
MMM-xiaomi.js:61 Starting module: MMM-xiaomi
main.js:508 All modules started!
(index):1 [Intervention] Slow network is detected. See https://www.chromestatus.com/feature/5636954674692096 for more details. Fallback font will be used while loading: http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Light.woff2
(index):1 [Intervention] Slow network is detected. See https://www.chromestatus.com/feature/5636954674692096 for more details. Fallback font will be used while loading: http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Bold.woff2
MMM-MotionDetector.js:72 MMM-MotionDetector: DiffCamEngine init successful.

rejas, 2020-12-09:
one more question before I have to go to sleep: how did you get the vcgencmd command running on macos catalina? this is used to turn the monitor off and on.
I always thought it was a tool only for raspbian?

3vidar, 2020-12-09:
Please see my first comment:
Turn on:
'caffeinate -u -t 1'
Turn off:
'pmset displaysleepnow'

isMonitorOn: function(resultCallback) {
	exec('pmset -g powerstate IODisplayWrangler | tail -1 | cut -c29'

this is what i changed in the node_helper.js to work for mac os

rejas, 2020-12-10:
Sorry, was late last night :-) Anyway, I added your commands to my local installation on my macbook. One line I had to change in isMonitorOn function in node_helper.js:
     
 resultCallback(out.includes("4"));

Now it seems to work here. Maybe you try that out?

3vidar, 2020-12-10:
@rejas thanks a lot for your help! Could you show me where your changed exactly the line? Sorry I‘m a absolute noob in coding! Thx

3vidar, 2020-12-11:
@rejas wowooo it works! Thanks a lot for your help. Could you mention here in github the Mac os settings? Just for others! Thank you again

rejas, 2020-12-11:
Thanks. I added a note to the README and hope to merge your code in the mainline later this year (or next ;-)

PR #41 (2020-12-11): Add info about mac support
Fixes #40
closed 2020-12-11 by rejas