rejas/MMM-MotionDetector #42: Motion detection pings network way too often makes everything lag bad

JoshuaShelswell, 2020-12-14:
can you make the interval it detects actual motion settable this would fix the issue

rejas, 2020-12-14:
I am confused, where does my module ping the netowrk?

rejas, 2020-12-14:
And what would a more easonable default value be in your opinion?

JoshuaShelswell, 2020-12-20:
sorry for the late reply, holiday stuff, sorry i wasnt that descriptive, yeah the captureIntervalTime was making the newsfeedticker lag pretty badly i set it to 3000 and it seems to be better on it

rejas, 2020-12-21:
Setting it to 3seconds would bean you'll  have to wait that long until you see an image (maybe a little more depending how quick your monitor turns on). 
Interesting though that my reuests and the newsfeedticker seem to lag. How does that manifest and what machine are you running the MM on?

JoshuaShelswell, 2020-12-21:
> 
> 
> Setting it to 3seconds would bean you'll have to wait that long until you see an image (maybe a little more depending how quick your monitor turns on).
> Interesting though that my code and the newsfeedticker seem to lag. How does that manifest and what machine are you running the MM on?

it actually makes everything lag, the most noticable things are the ticker and MMM-Globe (the fancy one with locations as it spins) im using a pi4 8gb, i noticed the slow speeds, i overclocked the pi to 2 ghz with @6v, very stable, have a 3A psu, and yeah i realize it would take 3 seconds to see you and then turn on the monitor but the camera has such a wide view and high sensitivity that it would turn on way before you were close to it anyway but regardless it still pings the pi way too often and at a rate that its noticeable, check out the network tab in the console in mm, it spams alot and it does make the already sortof laggy modules lag very badly

JoshuaShelswell, 2020-12-21:
[image]

rejas, 2021-01-24:
Sorry for the late replay @JoshuaShelswell holidays and such. Anyway, I am wondering why the MotionDetector Module should put anything on the network. Could you post a image of the network traffic you experience?

JoshuaShelswell, 2021-01-27:
if you go to the dev console and go to the network tab you can see it in action

rejas, 2021-01-31:
Ah, I see. Would you mind trying out the issue_42 branch to see if it reduces your network lag?

PR #44 (2021-02-12): Only update DOM when module is displayed
Fixes #42
closed 2021-02-12 by rejas