rejas/MMM-MotionDetector #43: MMM-Screencast and MMM-MotionDetector are incompatible

NicolasJorquera, 2021-01-19:
I have the MMM-MotionDetector which works just fine until i play a video with the MMM-Screencast module, when the screen turns off again after i play a video on youtube, it doesnt turn on again.
Someone had the same problem?
Sorry for my english.

NicolasJorquera, 2021-01-19:
it's like the camera will stop checking.

rejas, 2021-01-24:
Hi @NicolasJorquera are there any errors in the log / brwoser console? Could you post your config here so I can take a look myself?

NicolasJorquera, 2021-01-24:
Hi, im so sorry it didnt recognize as a code. Here it is

`* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
 address: "localhost",  // Address to listen on, can be:
       // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
       // - another specific IPv4/6 to listen on a specific interface
       // - "0.0.0.0", "::" to listen on any interface
       // Default, when address config is left out or empty, is "localhost"
 port: 8080,
 basePath: "/",  // The URL path where MagicMirror is hosted. If you are using a Reverse proxy
     // you must set the sub path here. basePath must end with a /
 ipWhitelist: [],  // Set [] to allow all IP addresses
               // or add a specific IPv4 of 192.168.1.5 :
               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
               // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

 useHttps: false,   // Support HTTPS or not, default "false" will use HTTP
 httpsPrivateKey: "",  // HTTPS private key path, only require when useHttps is true
 httpsCertificate: "",  // HTTPS Certificate path, only require when useHttps is true

 language: "es",
 logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
 timeFormat: 24,
 units: "metric",
 // serverOnly:  true/false/"local" ,
 // local for armv6l processors, default
 //   starts serveronly and then starts chrome browser
 // false, default for all NON-armv6l devices
 // true, force serveronly mode, because you want to.. no UI on this device

 modules: [
   {
   module: "MMM-Spotify",
   position: "top_bar",
   config: {
     debug: false,
   }
    },
  {
   module: "alert",
  },
  {
   module: "updatenotification",
   position: "top_bar"
  },
  {
   module: "clock",
   position: "top_left",
   config: {
    displaySeconds: false,
   }
  },
  {
     module: 'MMM-GmailFeed',
     position: 'top_left',
     config: {
   username: 'nicolas.jorquera@sansano.usm.cl',
   password: 'vzejsfdjrrnfrasu',
   updateInterval: 30000,
   maxEmails: 5,
   maxSubjectLength: 20,
   maxFromLength: 10,
   playSound: true,
   autoHide: true,
   displayMode: "table",
     }
    },
  {
     module: 'MMM-GmailFeed',
     position: 'top_left',
     config: {
   username: 'nicolas.jorqueramartinez@gmail.com',
   password: 'czgslwiwaiwtiuzm',
   updateInterval: 30000,
   maxEmails: 5,
   maxSubjectLength: 20,
   maxFromLength: 10,
   playSound: true,
   autoHide: true,
   displayMode: "table",
     }
    },
  {
     module: "currentweather",
     position: "top_right",
     config: {
   location: "Viña del Mar",
   locationID: "3868121", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
   appid: "78a8dbe67383d4d458b06c908f7b6a4f"
     }
    },
    {
     module: "weatherforecast",
     position: "top_right",
     header: "Pronóstico del clima",
     config: {
   location: "Viña del Mar",
   locationID: "3868121", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
   appid: "78a8dbe67383d4d458b06c908f7b6a4f"
     }
  },
  {
     module: "newsfeed",
     position: "bottom_bar",    // This can be any of the regions. Best results in center regions.
     config: {
   // The config property is optional.
   // If no config is set, an example calendar is shown.
   // See 'Configuration options' for more information.

   feeds: [
    {
     title: "Internacional",
     url: "https://www.elmostrador.cl/noticias/mundo/feed/",
    },
   {
     title: "Chile",
     url: "https://www.elmostrador.cl/destacado/feed/",
    },
    {
     title: "Ciencia",
     url: "https://www.investigacionyciencia.es/rss/noticias",
    },
    {
     title: "Tecnología",
     url: "https://es.digitaltrends.com/noticias/feed/",
    },

   ],
    showSourceTitle: true,
    showPublishDate: true,
    broadcastNewsFeeds: true,
    broadcastNewsUpdates: true,
    showDescription: true,
    updateInterval: 15000
     },
    },
    {
      module: 'MMM-Screencast',
      position: 'lower_third', // This position is for a hidden <div /> and not the screencast window
      config: {
    position: 'middle',
    height: 1050,
    width: 1280,
      }
     },
    {
     module: "MMM-MotionDetector",
     //position: "lower_third", // Optional. This can be any of the regions. Displays debug informations.
     config: {
    timeout: "60000",
     },
    },
 ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}`

rejas, 2021-01-24:
I deleted your config @NicolasJorquera because there were some passwords in it. Please repost without them. 
With the "< >" button in the texteditor you can mark i as code btw.

NicolasJorquera, 2021-01-24:
Oh thank you, i think now its ok.
And ive used the "< >" button, it doesnt recognize it as a code.

`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
 address: "localhost",  // Address to listen on, can be:
       // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
       // - another specific IPv4/6 to listen on a specific interface
       // - "0.0.0.0", "::" to listen on any interface
       // Default, when address config is left out or empty, is "localhost"
 port: 8080,
 basePath: "/",  // The URL path where MagicMirror is hosted. If you are using a Reverse proxy
     // you must set the sub path here. basePath must end with a /
 ipWhitelist: [],  // Set [] to allow all IP addresses
               // or add a specific IPv4 of 192.168.1.5 :
               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
               // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

 useHttps: false,   // Support HTTPS or not, default "false" will use HTTP
 httpsPrivateKey: "",  // HTTPS private key path, only require when useHttps is true
 httpsCertificate: "",  // HTTPS Certificate path, only require when useHttps is true

 language: "es",
 logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
 timeFormat: 24,
 units: "metric",
 // serverOnly:  true/false/"local" ,
 // local for armv6l processors, default
 //   starts serveronly and then starts chrome browser
 // false, default for all NON-armv6l devices
 // true, force serveronly mode, because you want to.. no UI on this device

 modules: [
   {
   module: "MMM-Spotify",
   position: "top_bar",
   config: {
     debug: false,
   }
    },
  {
   module: "alert",
  },
  {
   module: "updatenotification",
   position: "top_bar"
  },
  {
   module: "clock",
   position: "top_left",
   config: {
    displaySeconds: false,
   }
  },
  {
     module: 'MMM-GmailFeed',
     position: 'top_left',
     config: {
   username: '',
   password: '',
   updateInterval: 30000,
   maxEmails: 5,
   maxSubjectLength: 20,
   maxFromLength: 10,
   playSound: true,
   autoHide: true,
   displayMode: "table",
     }
    },
  {
     module: 'MMM-GmailFeed',
     position: 'top_left',
     config: {
   username: '',
   password: '',
   updateInterval: 30000,
   maxEmails: 5,
   maxSubjectLength: 20,
   maxFromLength: 10,
   playSound: true,
   autoHide: true,
   displayMode: "table",
     }
    },
  
  {
     module: "currentweather",
     position: "top_right",
     config: {
   location: "Viña del Mar",
   locationID: "3868121", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
   appid: ""
     }
    },
    
    {
     module: "weatherforecast",
     position: "top_right",
     header: "Pronóstico del clima",
     config: {
   location: "Viña del Mar",
   locationID: "3868121", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
   appid: ""
     }
  },
  {
     module: "newsfeed",
     position: "bottom_bar",    // This can be any of the regions. Best results in center regions.
     config: {
   // The config property is optional.
   // If no config is set, an example calendar is shown.
   // See 'Configuration options' for more information.

   feeds: [
    {
     title: "Internacional",
     url: "https://www.elmostrador.cl/noticias/mundo/feed/",
    },
   {
     title: "Chile",
     url: "https://www.elmostrador.cl/destacado/feed/",
    },
    {
     title: "Ciencia",
     url: "https://www.investigacionyciencia.es/rss/noticias",
    },
    {
     title: "Tecnología",
     url: "https://es.digitaltrends.com/noticias/feed/",
    },

   ],
    showSourceTitle: true,
    showPublishDate: true,
    broadcastNewsFeeds: true,
    broadcastNewsUpdates: true,
    showDescription: true,
    updateInterval: 15000
     },
    },
    {
      module: 'MMM-Screencast',
      position: 'lower_third', // This position is for a hidden <div> and not the screencast window
      config: {
    position: 'middle',
    height: 1050,
    width: 1280,
      }
     },
    {
     module: "MMM-MotionDetector",
     //position: "lower_third", // Optional. This can be any of the regions. Displays debug informations.
     config: {
    timeout: "60000",
     },
    },
 ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
</div>`

rejas, 2021-02-01:
Just taking your latest config example I get this error on the console:

[01.02.2021 09:01.25.945] [ERROR] MMM-Screencast:CONFIG-ERROR: There was an error with your positioning config. Please check your config.

Please check your config again....