rejas/MMM-MotionDetector #45: Another module that requires MagicMirror address to be 0.0.0.0

NicolasJorquera, 2021-02-08:
I know that is a segment that indicates how to set up a reverse proxy for the other module, but i have two questions.
1. How to install http-proxy-middleware
2. The routes.json and proxyserver.js files, where they should be added?

rejas, 2021-02-09:
Maybe @bavario can help you there, since he wrote that part of the README?

bavario, 2021-02-11:
Hi @NicolasJorquera 

you can install the middleware like any other npm package (npm i http-proxy-middleware). 

I'm not completely sure where I had put those files but it doesn't really matter. All you have to do is to ensure you can start the express server with an npm start script or even better with PM2 to ensure it will be restarted if you reboot your PI. I'd suggest to put those things anywhere within your MM installation eg a subfolder within modules.

Currently my Mirror and Pi are not in use due to changing flats. But if you need more instructions please let me know, than I can grap my PI and look to it in detail.

Regards

rejas, 2021-02-22:
Did that help you @NicolasJorquera  ?

NicolasJorquera, 2021-02-22:
I just change the remote-control module for the telegram bot one with google assistant, and works just fine.
Thank you a lot but i didnt have the chance to try it out

closed 2021-02-23 by rejas