rejas/MMM-MotionDetector #47: Multi-Camera issue

camaro4life18, 2021-02-10:
Hello, I am trying to run MM with Motion in the background. I have 2 camera's connected to my Pi, one is configured to Motion, the other is meant for MMM-MotionDetector, however I can't get it to recognize a different camera. It keeps defaulting to the camera that is already in use and throws an error...
NotReadableError: could not start video source

camaro4life18, 2021-02-10:
As usual, I got impatient and solved my own problem by changing the code to be this...
 navigator.mediaDevices
      .getUserMedia({ video: { deviceId: "video0"} })
Basically had to hard code it to pick a specific camera. If I get time I may put in a pull request to add this as a config option. The weird behavior however, is It acts like it is using both cams now for motion detection.

rejas, 2021-02-13:
Hi @camaro4life18 I also got impatient and crated a PR with the option to specify the deviceId. Can you give it a try and report back if it helps you?

closed 2021-02-24 by rejas