rejas/MMM-MotionDetector #49: Cannot use/load this module

tcsabina, 2021-02-28:
Hi,

Just wanted to experiment with this module, using a Trust gxt 1160 vero USB webcam.

Upon adding this module into the config, MM just stays black.
Error log content:
`WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Bind must be called on a function
App threw an error during load
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:25:38
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:27:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:29:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:25:38
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:27:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:29:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
/proc/self/exe: /lib/arm-linux-gnueabihf/libdbus-1.so.3: no version information available (required by /proc/self/exe)

`

The config.js section:
`                {
                        module: "MMM-MotionDetector",
                        position: "top_left", // Optional. This can be any of the regions. Displays debug informations.
                        config: {
                                captureIntervalTime: 1000,
                                scoreThreshold: 20,
                                timeout: 120000,
                        }
                },
`

Any tips?

rejas, 2021-02-28:
What version of the MM core are you using? And what platform is it running on?

tcsabina, 2021-03-01:
Hi rejas,

This is an old Raspberry Pi: 2B Model (Revision a01041)
With an old MM: 2.7.1 (based on the CHANGELOG.md)

Time to upgrade? At least MM?

rejas, 2021-03-01:
Upgrading MM to the latest version could help. My module uses some logger functions that were definitly added later than 2.7.1.

tcsabina, 2021-04-14:
Hi rejas,

Updated to latest MM, and it works fine!

Actually I wiped my RPi SD card, and installed the latest Raspbian (buster), with the latest MM.

1 question:
what does to module call when turning the monitor off/on? vcgencmd, tvservice, or something else?

Regards

closed 2021-04-14 by tcsabina

rejas, 2021-04-14:
Glad it worked out.
To answer your question: vcgencmd :-)