rejas/MMM-MotionDetector #50: motion not detected after last MM update

freesty82, 2021-04-06:
hi,
i have updated MM to version 2.15.0 and motionDetector stop working. The module starts but motion is no longer detected, so the monitor turns off and never turns on again. 
I've the last MMM-MotionDetector version and a pi camera.

Thanks

rejas, 2021-04-06:
Hi @freesty82 what version is your MotionDetector module?

freesty82, 2021-04-06:
i've the latest version, i already try to reinstall motionDetector with no success

the logs seems fine:

0|MagicMirror  | > magicmirror@2.15.0 start /home/pi/MagicMirror
0|MagicMirror  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|MagicMirror  | [06.04.2021 23:38.55.684] [LOG]   
0|MagicMirror  | Starting MagicMirror: v2.15.0
0|MagicMirror  | [06.04.2021 23:38.55.702] [LOG]   
0|MagicMirror  | Loading config ...
0|MagicMirror  | [06.04.2021 23:38.55.727] [LOG]   
0|MagicMirror  | Loading module helpers ...
0|MagicMirror  | [06.04.2021 23:38.55.737] [LOG]   
0|MagicMirror  | No helper found for module: alert.
0|MagicMirror  | [06.04.2021 23:38.55.758] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:38.55.764] [LOG]   
0|MagicMirror  | Module helper loaded: mmpm
0|MagicMirror  | [06.04.2021 23:38.55.794] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:38.55.796] [LOG]   Module helper loaded: MMM-PiTemp
0|MagicMirror  | [06.04.2021 23:38.56.190] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:38.58.714] [LOG]   Module helper loaded: MMM-TelegramBot
0|MagicMirror  | [06.04.2021 23:39.02.347] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.02.348] [LOG]   Module helper loaded: MMM-GoogleAssistant
0|MagicMirror  | [06.04.2021 23:39.04.477] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.04.479] [LOG]   Module helper loaded: MMM-Assistant2Display
0|MagicMirror  | [06.04.2021 23:39.04.877] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.04.878] [LOG]   Module helper loaded: MMM-homeassistant-sensors
0|MagicMirror  | [06.04.2021 23:39.04.881] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.04.889] [LOG]   Module helper loaded: MMM-MotionDetector
0|MagicMirror  | [06.04.2021 23:39.05.879] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.05.881] [LOG]   Module helper loaded: MMM-Tado
0|MagicMirror  | [06.04.2021 23:39.06.895] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.06.902] [LOG]   
0|MagicMirror  | Module helper loaded: MMM-UpdateNotification
0|MagicMirror  | [06.04.2021 23:39.06.904] [LOG]   No helper found for module: clock.
0|MagicMirror  | [06.04.2021 23:39.07.193] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.07.195] [LOG]   
0|MagicMirror  | Module helper loaded: MMM-Remote-Control
0|MagicMirror  | [06.04.2021 23:39.07.595] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.07.597] [LOG]   Module helper loaded: calendar
0|MagicMirror  | [06.04.2021 23:39.07.599] [LOG]   No helper found for module: MMM-Globe.
0|MagicMirror  | [06.04.2021 23:39.07.602] [LOG]   No helper found for module: compliments.
0|MagicMirror  | [06.04.2021 23:39.08.390] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.08.391] [LOG]   Module helper loaded: MMM-Weather
0|MagicMirror  | [06.04.2021 23:39.09.245] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.09.247] [LOG]   Module helper loaded: MMM-AVStock
0|MagicMirror  | [06.04.2021 23:39.10.420] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [06.04.2021 23:39.10.425] [LOG]   Module helper loaded: MMM-NewsFeed
0|MagicMirror  | [06.04.2021 23:39.10.427] [LOG]   All module helpers loaded.
0|MagicMirror  | [06.04.2021 23:39.10.646] [LOG]   
0|MagicMirror  | Starting server on port 8080 ... 
0|MagicMirror  | [06.04.2021 23:39.10.661] [LOG]   
0|MagicMirror  | Server started ...
0|MagicMirror  | [06.04.2021 23:39.10.665] [LOG]   
0|MagicMirror  | Connecting socket for: mmpm
0|MagicMirror  | [06.04.2021 23:39.10.669] [LOG]   
[... 153 lines cut ...]
0|MagicMirror  |     - MMM-Tado: Not all data present
0|MagicMirror  | [06.04.2021 23:39.35.549] [LOG]   
0|MagicMirror  |     - MMM-Tado: Checking if all data is present
0|MagicMirror  | [06.04.2021 23:39.35.555] [LOG]       - MMM-Tado: Not all data present
0|MagicMirror  | [06.04.2021 23:39.36.048] [LOG]   
0|MagicMirror  |     - MMM-Tado: Got zones()
0|MagicMirror  | [06.04.2021 23:39.36.560] [LOG]   
0|MagicMirror  |     - MMM-Tado: Checking if all data is present
0|MagicMirror  | [06.04.2021 23:39.36.569] [LOG]       - MMM-Tado: Not all data present
0|MagicMirror  | [06.04.2021 23:39.37.606] [LOG]   
0|MagicMirror  |     - MMM-Tado: Checking if all data is present
0|MagicMirror  | [06.04.2021 23:39.37.607] [LOG]       - MMM-TADO: All data is present
0|MagicMirror  | [06.04.2021 23:40.23.603] [LOG]   
0|MagicMirror  | <Buffer 33 38 2e 30 0a>

after 6 minutes:
0|MagicMirror  | [06.04.2021 23:45.20.596] [INFO]  
0|MagicMirror  | MMM-MotionDetector: deactivating monitor.
0|MagicMirror  | [06.04.2021 23:45.20.743] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor status is display_power=1
0|MagicMirror  | [06.04.2021 23:45.21.016] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor has been deactivated.
0|MagicMirror  | [06.04.2021 23:45.21.565] [LOG]   

the display says always "score 0", so seems that can't detect motion anymore.

rejas, 2021-04-08:
mhh.... like the saying goes: it works on my side. maybe some other modules interfere now? can you disable all other modules and see if it works then? and can you psot your config (the relevant parts) ?

freesty82, 2021-04-10:
Hi,
thank for you reply. I disabled all other modules with no luck, 
I tried another module (MMM-selfieShot) to see if the camera works, and it has no problems.

thanks

this is the log and my config:

0|MagicMirror  | > magicmirror@2.15.0 start /home/pi/MagicMirror
0|MagicMirror  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|MagicMirror  | [10.04.2021 09:30.21.821] [LOG]   
0|MagicMirror  | Starting MagicMirror: v2.15.0
0|MagicMirror  | [10.04.2021 09:30.21.827] [LOG]   
0|MagicMirror  | Loading config ...
0|MagicMirror  | [10.04.2021 09:30.21.835] [LOG]   
0|MagicMirror  | Loading module helpers ...
0|MagicMirror  | [10.04.2021 09:30.21.867] [LOG]   
0|MagicMirror  | Initializing new module helper ...
0|MagicMirror  | [10.04.2021 09:30.21.868] [LOG]   
0|MagicMirror  | Module helper loaded: MMM-MotionDetector
0|MagicMirror  | [10.04.2021 09:30.21.870] [LOG]   
0|MagicMirror  | All module helpers loaded.
0|MagicMirror  | [10.04.2021 09:30.22.017] [LOG]   
0|MagicMirror  | Starting server on port 8080 ... 
0|MagicMirror  | [10.04.2021 09:30.22.035] [LOG]   
0|MagicMirror  | Server started ...
0|MagicMirror  | [10.04.2021 09:30.22.039] [LOG]   
0|MagicMirror  | Connecting socket for: MMM-MotionDetector
0|MagicMirror  | [10.04.2021 09:30.22.110] [LOG]   
0|MagicMirror  | Sockets connected & modules started ...
0|MagicMirror  | [10.04.2021 09:30.22.722] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor status is display_power=0
0|MagicMirror  | [10.04.2021 09:30.22.727] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor is OFF.
0|MagicMirror  | [10.04.2021 09:30.22.783] [LOG]   
0|MagicMirror  | Launching application.
0|MagicMirror  | [10.04.2021 09:30.23.077] [ERROR] 
0|MagicMirror  | (node:23902) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
0|MagicMirror  | [10.04.2021 09:30.25.853] [INFO]  
0|MagicMirror  | MMM-MotionDetector: activating monitor.
0|MagicMirror  | [10.04.2021 09:30.25.901] [INFO]  MMM-MotionDetector: monitor status is display_power=0
0|MagicMirror  | [10.04.2021 09:30.25.942] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor has been activated.
0|MagicMirror  | [10.04.2021 09:31.02.048] [INFO]  
0|MagicMirror  | MMM-MotionDetector: deactivating monitor.
0|MagicMirror  | [10.04.2021 09:31.02.103] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor status is display_power=1
0|MagicMirror  | [10.04.2021 09:31.02.145] [INFO]  
0|MagicMirror  | MMM-MotionDetector: monitor has been deactivated.

this is my config:

var config = {
    address: "localhost",
	//address: "0.0.0.0", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "0.0.0.0", "::" to listen on any interface
	electronOptions: {
        webPreferences: {
            webviewTag: true
        }
    },                      // Default, when address config is left out or empty, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1","192.168.1.136", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.0/28"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "it",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device

	modules: [

        {
		   module: "MMM-MotionDetector",
            disabled: false,
		    position: "bottom_right",	// Optional. This can be any of the regions. Displays debug informations.
		    config: {
			 //The config property is optional.
			 //See 'Configuration options' for more information.
                 timeout: 36000

		    }
	    }

            
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

rejas, 2021-04-10:
Unfortunalty I dont have a PI camera so I cant test if something is wrong with that since v2.15

Can you start your MM in the dev mode (`npm run start:dev` instead of the normal npm run start) and look for errors in the devtools frame?

frosty-snow, 2021-04-19:
@freesty82 
Did you ever find a workaround to this problem? I'm also trying to get this setup on my pi and have confirmed it is not the camera as I can capture images with it. Like you describe above, I'm having the same behavior of motion not being detected. MM and this module all up to date (this module just installed today for the first time).

1morebikeride, 2021-08-19:
Still having this issue.  Did anyone ever find a solution?

rejas, 2021-08-20:
hi @1morebikeride I couldnt reproduce this issue on my local setup where my detector runs fine. Can you post your config and setup? What does your log say?