rejas/MMM-MotionDetector #52: Deactivates monitor but will not wake up

cdr93, 2021-11-12:
Hello,

I want to use this module but I have a problem. Indeed, the module deactivates the screen but it does not reactivate it.
My screen is a television connected to the PI by an HDMI cable. I am using CEC Utils to be able to manage the screen with crontab. The commands that are working to turn on and off the screen are:

  echo on 0 | cec-client -s -d 1 == Turn on the screen
  echo standby 0 | cec-client -s -d 1 == Turn off the screen

How could I integrate these commands in your module?

Thanking you in advance,

Regards,

rejas, 2021-11-15:
Hi @cdr93 thanks for you rinterest in my project. I (re-)started work on a branch to integrate different utils for monitor handling.

If you want you can checkout the branch "mac" for my latest work there. Alas, I would need your help to implement the status call for cec-client:
What would be the ouptut of

echo 'pow 0' | cec-client -s -d 1

when the monitor is on and when it is off?

Thanks for your help

cdr93, 2021-11-21:
Hello,
Thank you for your interest in my question.
When I type the screen on command:
`pi@raspberrypi:~ $ echo 'pow 0' | cec-client -s -d 1
opening a connection to the CEC adapter...
power status: on
pi@raspberrypi:~ $
`
When I type the screen off command:
`pi@raspberrypi:~ $ echo 'pow 0' | cec-client -s -d 1
opening a connection to the CEC adapter...
power status: standby
pi@raspberrypi:~ $
`
Hoping to have answered your request,
Regards,

rejas, 2021-11-28:
Thx for the input, I added it to the branch "mac". Can you check that out and test it to see if it works?

cdr93, 2021-11-29:
Hello,
I still have the same problem. The screen is indeed deactivated by the module but it does not reactivate it. I have to turn it back on manually.
Regards