rejas/MMM-MotionDetector #53: error calling monitor status: /bin/sh: 1: vcgencmd: not found

titouan73, 2021-11-12:
Hello firstly congrats for this module, I'm currently trying to use an iMac 27" running on linux distro (Pop_OS which is Ubuntu basically) and I got this error causing my monitor is never going black : 
```
[12.11.2021 15:38.04.598] [ERROR] MMM-MotionDetector: error calling monitor status: /bin/sh: 1: vcgencmd: not found
```
Those error happens every time the module try to activate or deactivate the monitor.
I've read online that vcgencmd is a package only available on Raspberry PI.
Hove you in mind a solution or an alternative package that would work on other distro that raspPI ? 

Thank you

rejas, 2021-11-15:
Hi @titouan73 I (re-)started work to make multiple utils feasible in the MMM-module. 

The only two options right now in my branch "mac" are mac-os utils and cec-utils (as mentioned here https://github.com/rejas/MMM-MotionDetector/issues/52)

Not sure what options are on your Pop_os, that is very distro specific :-(

titouan73, 2021-11-15:
It's a gnome based distro I think we can modify the code to wake up the screen with a command ?

rejas, 2021-11-15:
sorry, dont have the time to google or install that myself to see whats possible. If you can find the info on how to de/activate and get the status for your monitor I am happy to incorporate it.

titouan73, 2021-11-15:
Thanks a lot I'll do that

titouan73, 2021-11-15:
Ok the command is `xset dpms force off` to turn off the screen, to turn it on you've just to change off to on. Some people advice to put `sleep 1; xset dpms force off `. Check https://superuser.com/questions/374637/how-to-turn-off-screen-with-shortcut-in-linux for more info. Thank again for you help