rejas/MMM-MotionDetector #55: Cannot use Pi camera for this module, but followed the documentation

Jochbart, 2022-01-29:
I changed the module file, but I get an error now when rebooting the pi:
[image]
[image]

Please help.

rejas, 2022-01-29:
Sorry, cant help you directly since I dont own a pi camera. But what does your output say when running 
"systemctl status systemd-modules-load.service"
as mentioned in the error log above?

Jochbart, 2022-01-29:
[image]

Jochbart, 2022-01-29:
[image]

rejas, 2022-01-29:
Looks like you got a typo there: you wrote v412 but it should be v4l2

Jochbart, 2022-01-31:
Hi rejas,

sorry, my bad, you were right. The rebooting is working now, but still get an error in MM:
[image]

rejas, 2022-02-05:
Can confirm that happens on a fresh installation of raspi OS starting with their latest releases. You need to enable the legacy camera support in the raspi-config tool.

I updated the README accordingly. Can you confirm that it solves your problem?

Jochbart, 2022-02-08:
Hi rejas,
I did it, but same error.

rejas, 2022-03-02:
Hi guys, just found out that there is now a "legacy" raspberry os, which uses the old debian. Maybe you can check this out? https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

fernandosalomao, 2022-03-02:
Looks like there is an issue with chromium since 89.0.4389.128
https://stackoverflow.com/questions/69300651/electron-rpi-requested-device-not-found-camera

vijaybala1, 2023-06-30:
Is this broken ? Or does it work with latest version of MM ? Using PI 4 with PI camera.

rejas, 2023-07-01:
> Is this broken ? Or does it work with latest version of MM ? Using PI 4 with PI camera.

Its working for me with latest version of MM when using my PS3 camera.

I dont have the time at the moment to test it with the latest raspiOS release (the underlying operating system) and a PI camera. Any help would be appreciated.

vijaybala1, 2023-07-01:
Happy to help. Its not working with Pi-Camera ( V2) . I also have a standard 720 Webcam (MIcrosoft Brand) that also not work. I get a "NotfoundError" . Happy to help debug .