rejas/MMM-MotionDetector #56: Error: getUserMedia is not implemented in this browser

iTDB-Gent, 2022-02-03:
I reinstalled raspbian and now keep getting following message:
Error: getUserMedia is not implemented in this browser

Tested older and new versions of raspbian on Pi4, also Stretch version on Pi3.
Config > address: “0.0.0.0”,
When changing to “localhost” or “127.0.0.1” you get the “NotFoundError: Requested device not found”

https://forum.magicmirror.builders/topic/16315/mmm-motiondetector-error-getusermedia

rejas, 2022-02-05:
Could you try enabling the legy camera support in the raspi-config tool?

3 Interface Options -> I1 Legacy Camera

iTDB-Gent, 2022-02-06:
The Legacy Camera is on. 
Run as server and start chromium > authorize the camera. And it works

jacobpearson, 2022-04-18:
I'm having the same issue too, runs fine in Chromium once authorised. However the pi goes unresponsive overnight using this method (still investigating)

rejas, 2022-04-23:
FYI: I switched back to the legacy raspios (https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/) and there the bug doesnt seem to appear.

iTDB-Gent, 2022-04-25:
Correct. On the legacy raspios it works. 
Thx

rejas, 2022-04-27:
So we can close thi sissue for now?

misk0, 2022-05-20:
> The Legacy Camera is on. Run as server and start chromium > authorize the camera. And it works

I have the same issue. I'm using MM more than 4 years and now I just updated MM to version 2.0.19. After the update MotionDetector stop to work with message "Requested device not found". 

I tried workaround (start MM with 'npm run server' and open chromium thru remote XDisplay (MobTerm). First time it asked me to allow camera and next time it didn't. 
But when I close all and run MM from pm2 it shows me the same error message.

Is it possible to make this chromium setting permanent for all browser session, or something similar?

My mirror is in my living room and I don't /can't have mouse and keyboard always connected to start it manually every time.

iTDB-Gent, 2022-10-31:
It looks like this won't work again with the latest version of MagicMirror.
When I look at Chromium, you can activate the camera on localhost & 127.0.0.1 (permission for this site) but on 0.0.0.0 it is grayed out. Think this will also be the case for Electronjs.
Only possibility I see is;
- make Electronjs authenticate the camera when starting module
- make Electronjs authenticate the camera when starting MagicMirror
  > set camera permission for 0.0.0.0

kevinoue, 2023-01-18:
Has anyone had any luck with this?