rejas/MMM-MotionDetector #58: Problem

1eretoile, 2023-06-30:
[image]
[image]
Hello
Je n'arrive pas à le faire fonctionner alors que j'ai bien ma caméra pi qui fonctionne

rejas, 2023-07-01:
sorry i dont speak french. assuming google translate is correct, you got problems with a pi camera? Again sorry, I dont have one myself to test this module against it.
What OS are you using?

1eretoile, 2023-07-01:
Oui
Désolé je parle pas anglais
Je suis bie' en localhost, j'ai paramétré comme @rev138
Et j'ai bien ma caméra pi qui marche sans passer par magic mirror
J'ai installer le modules
Recopier la config dans config. Js

J'ai dernière version debian