rejas/MMM-MotionDetector #60: Screen does not power off

vavida52, 2024-03-06:
Hello,

I started to reuse my magic mirror after like 2 years or so. Back then when I used the MM this module worked perfectly fine (using the PS camera). 
I recently had to rebuild my MM because my SD card died and I try to set this module up again and it seems to detect properly. The issue is that the screen does not turn off although it is shown that the screen was shut off. Is it somehow related to power settings of the pi?

Thank you in advance for any tip.

vavida52, 2024-03-08:
Solved my issue. Had to change the boot/config.txt

closed 2024-03-08 by vavida52

BGH-H, 2025-10-31:
im  having the same problem, the modules loads fine, motion is detected, the log states that monitor is powering off, but the screen stays on..... what exactly did you change in config.txt? Im running on a usb webcam currently (logitec). 

(I also am doing a rebuild since mine became too slow, which is very difficult to pull of since modules are nog updated and everything is running on never npm/dependancies etc).